laurentalacoque/TouchSelfie-extended #30: stokage photo

pikeupe, 2018-07-10:
Bonjour

 J’espère demander au bon endroit, comment peut-on faire pour avoir les photos de meilleurs taille sur le support USB car actuellement les photos sur l'USB ne dépassassent pas 250 ko

comme je débute pas évident merci pour ce travail

laurentalacoque, 2018-07-11:
Hello @pikeupe,
Sorry for my translation, english is preferred for international collaboration ;)

## Short answer
Tune the `snap_size` property of the `EFFECT_PARAMETERS` dict in file `scripts/constants.py` to enhance snaphot resolution. (You will still be limited to your camera hardware)

## Longer answer
Picture sizes depends on: 
- Snapshot resolution (and utlimately camera resolution)
- Image content
- JPEG compression

Using PiCamera, you can only act on snapshot resolution.
I chose to keep resolution low (although higher than original TouchSelfie) in order to obtain lightweight yet readable snapshots.

Typically, the settings in `constants.py` are tuned to the quarter of picamera v2.1 full resolution.
You can change the `snap_size` values up-to the full camera resolution (which is (3280,2464) for the official v2.1 camera module)

laurentalacoque, 2018-07-11:
# How to solve

- Change `snap_size` properties in `EFFECT_PARAMETERS` dict of `constants.py`

- Potential enhancement: add a quality/size configuration option
  - [ ] Detect camera max resolution
  - [ ] Ask for user's preferences in the setup step
  - [ ] Ajust "snap_size" properties accordingly

closed 2018-07-11 by laurentalacoque

pikeupe, 2018-07-12:
Hello

thank you for your explanation and this work