laurentalacoque/TouchSelfie-extended #38: Unknown pull_up_down mode 0

zorruno, 2018-07-13:
After following standard install setup, I get this error when running ./photobooth.sh
Error included a Traceback, but the issue was in hardware_buttons.py Line 35

ValueError: `Unknown pull_up_down mode 0`

I don't have any hardware buttons connected at the moment, but if I put an actual value for the mode somewhere in hardware_buttons.py (eg `mode = "pull_down"` added around line 45) everything works fine so I'm not sure at the moment why it isn't pulling this from constants.py - it is correct in there, but is set to a default of 0 in hardware_buttons.py... hence the error.

I'll have a look at this more when I have added some hardware buttons at some stage - I guess I could have equally set `RPI_GPIO_EXISTS = False` in the mean time.

laurentalacoque, 2018-07-13:
Hey @zorruno ;)

It seems that the default value for systems without a GPIO was incorrect!

- This bug shouldn't show-up in normal condition (when `hardware_buttons` module is called from within photobooth script)

I just pushed a quickfix in master branch, could you please `git pull` and tell me if it's fixed?

Solved in [this commit](https://github.com/laurentalacoque/TouchSelfie/commit/78f6aa200773239fbf57cf8ea7f8830904a61608)

zorruno, 2018-07-14:
yes fixed in master now thanks, but does that mean the values aren't being pulled from the constants file?

I haven't tried connecting any i/o yet, but it looks like the software buttons can be turned on/off fine in configuration.json