laurentalacoque/TouchSelfie-extended #40: error on USB

pikeupe, 2018-07-27:
Hello
I just made the update everything works fine except when I want to saved the photos on the USB key
he tells me:

Traceback (most recent call last):
  File "user_interface.py", line 481, in snap
    os.rename(self.last_picture_filename, new_filename)
OSError: [Errno 18] Invalid cross-device link

thank you for your work and help

laurentalacoque, 2018-07-29:
Hi @pikeupe 
Good one!
This is an unforeseen limitation of rename. I'll address this when I'm back from holidays next week 😉

laurentalacoque, 2018-07-29:
# todo
  [ ] change os.rename to shutil.copy and os.remove

pikeupe, 2018-07-30:
Hello

thank you and have a good holiday

laurentalacoque, 2018-08-06:
Problem solved in merge #43 
Now available in master.

closed 2018-08-06 by laurentalacoque