laurentalacoque/TouchSelfie-extended #44: Print function and effects

Pikokosan, 2018-08-07:
Hi! So we used the original version of this software for my fiancee's wedding shower over the weekend. I was presented with some things that needed to be done better. 

I found your version and its infinitely better.

2 things that I'll be trying to work on are.

#1 adding a button to send photos directly to a cannon selphy cp1300 from the raspberry.

#2 visual effects. Picamera has a image_effects function that has some predefined effects. I would like users to be able to view and pick from them if they wanted. This function is very low on my list.

If anyone has any insight into this help is appreciated. Ofcourse I'll share what I learn here. 

Thanks
  -Pikoko

laurentalacoque, 2018-08-07:
Hey @Pikokosan 
Thanks for your comment !

- For 1/ do you know if there's a driver for the selphy on raspian?
- For 2/ Nice idea, I'll think about it ;)

Pikokosan, 2018-08-08:
from what i've been seeing it is possible using CUPS. I found this project https://github.com/zoroloco/boothy/ which says he uses a cp1200. I should know more later this evening.

I saw your last merge! thank you. I plan to put the photos on a flash drive

laurentalacoque, 2018-08-12:
Printer support added in merge #46 
Thanks @Pikokosan !

laurentalacoque, 2018-08-14:
# Addressing builtin picamera effects
This is a work in progress in branch [builtin-effects](https://github.com/laurentalacoque/TouchSelfie/tree/builtin-effects)

laurentalacoque, 2018-08-14:
Builtin effects support now proposed in Pull Request #47

commit 025444ec (2018-08-16): Image Effects

Support of PiCamera builtin image effects
This definitely closes #44
closed 2018-08-16 by laurentalacoque