laurentalacoque/TouchSelfie-extended #48: Ajouter le logo.png

karlzre, 2018-09-13:
Bonjour Laurent, 

Je viens d'installer et tester ton photobooth, c'est génial !!!

Je ne trouve juste pas comment mettre le logo.png sur les photos, d'apres le readme cela devrait être pendant le setup.sh, mais je n'ai pas l'option.

J'ai aussi changé dans configuration.json le logo est "null", j'ai essaye de le changer en mettant logo.png mais le script ne démarre plus.

Peux tu m'indiquer ou le définir ?

Merci d'avance.

laurentalacoque, 2018-09-15:
Will take a look into this ;)

laurentalacoque, 2018-09-16:
I should (really) add this in setup, but you could try to 
- put your logo file in the `scripts` directory (e.g. `scripts/mylogo.png`)
- change the `configuration.json` `logo_file` key to something like this: `"logo_file":"mylogo.png",`
- relaunch the photobooth in the `scripts` directory using `python user_interface.py -df` (this will disable fullscreen mode
- copy/paste any error that cause the crash in this issue report

karlzre, 2018-09-19:
Hi laurent, 

This looks to works

laurentalacoque, 2018-09-19:
Nice!
Closing the bug then.

closed 2018-09-19 by laurentalacoque