laurentalacoque/TouchSelfie-extended #49: Exit touchSelfie

karlzre, 2018-09-13:
Bonjour Laurent, 

Peux tu m'indiquer comment quitter TouchSelfie ?

Avec un clavier la touche Esc fait l'affaire mais sans clavier, tu indique un long press, mais cela ne semble pas fonctionner.

Merci

laurentalacoque, 2018-09-15:
Hey @karlzre!
Thanks for your input.

Could you please try harder and really press Touchselfie's background during at lease 3 seconds?
When you lift your finger after a (very) long press, a dialog should appear from which you can choose to exit the application

karlzre, 2018-09-19:
Iv' try a long press more tahn 3 sec, on different places in touch selfie but didn't work.

Maybe a misconfiguration or issue on my touchscreen

laurentalacoque, 2018-11-05:
Unable to reproduce the bug, I'm closing it.
Feel free to reopen

closed 2018-11-05 by laurentalacoque