laurentalacoque/TouchSelfie-extended #54: control arduino for led flash

karlzre, 2018-10-25:
hi Laurent 

i Will place some led to make some animation on the selfiebox.

do you think  it should be possible To send a command from the pi to the arduino trough the gpio so i can make the led switch white when the photo is taken? 

thx

laurentalacoque, 2018-10-27:
Hi @karlzre :)
It's possible!
There's a  [__countdown_set_led](https://github.com/laurentalacoque/TouchSelfie-extended/blob/3c80b805d555dd1f3d99321a1ffe9af8e32931e6/scripts/user_interface.py#L561) in `user_interace.py` that blinks before the snapshot.
Currently it does nothing but blink the picamera led, but you can add your GPIO code here.

For a flash feature, you might want to light the flash somewhere in `__show_countdown` function and set it off just after the camera.capture (warning, one instance per effect) in the `user_interface` `snap` function

karlzre, 2018-11-04:
Thank you Laurent that works !!!

i'm sending a "variable value" trough serial to the arduino .

karlzre, 2018-11-04:
one more question, actually i only switch the flash on  in __show_countdown by sending a serial command to the arduino.

If i would like to send another serial command during the count down to "write" the countdown on a neopixel. for example when the 5 appear on the screen i would like to write 5 with the leds also, when the 4 appear, i would write the 4 etc...

Each number will be a different serial command, where can i add them ?

Thx

laurentalacoque, 2018-11-05:
Hey ;)
You can do this in `user_interface.py` -> `__show_overlay_countdown`
In this function, around line 648 you have the countdown loop that is run every seconds
Just add your code line 649 and you should be good

hint: the expression `(countdown -i)` evaluates to the number of seconds remaining

closed 2018-11-05 by laurentalacoque

karlzre, 2018-11-19:
Hi Laurent,

I've add this:
#what overlay image to select:
            ser.write(str(i))
            overlay_image = None

I can then blink a count down on a neopixe led strip when i take one
picture .
But when i take the option for 4 pictures i have the correct count down for
the first picture but for the next three it show the countdown 5 on the led
for the value 3 on the screen. i think i has a value of 0 this run the case
0 in my arduino script.

We have played with some friends and we think it should be better to have 5
sec between the next photo when we choose the 4 pictures.( actually it's 3)

Where can i change this behaviour ?

Thx a lot again for your help, will send you a video as soon as it's
finished -)

karlzre, 2018-11-19:
Hi Laurent, Found ;-)

changed config.countdown2 to config.countdown1 ;-)

karlzre, 2018-11-19:
hi Laurent
have à look at this.
https://youtu.be/RaZcAe4kcLQ

thank you again !

laurentalacoque, 2018-11-21:
Very nice!
Thanks for the video 😉

karlzre, 2018-12-01:
Hi Laurent,

It looks like my Pi has been hacked, my operator complain that something
send ssh connection from my internet acces.

A wireshark show a lot of ssh sessions from my pi to somwhere.

Did the photobooth use ssh to load pictures to google ?

If no, could you confirm i can back up the photobooth just by copying the
photobooth folder and copy it to a fresh pi install ?

Thx

laurentalacoque, 2018-12-01:
Hello,
No, the photobooth program doesn't use ssh at all.
All it does is to connect to google (for mail or upload) via https

Yes, you can zip the photobooth directory and unzip it on a fresh pi. You will keep your configuration and google access token.
Make sure the path to the local storage is still good on the fresh install

karlzre, 2019-01-19:
hi Laurent

i actually in a wedding and i cet the issue attached.

do you know if it is easy to fix ?

laurentalacoque, 2019-01-20:
Sorry I don't understand what issue, can you elaborate ?

karlzre, 2019-01-21:
My Pi was hacked and was trying to scan for others PI.

Issue fixed after format and reinstall, nothing linked with Touchselfie