laurentalacoque/TouchSelfie-extended #55: Pulling list of Google Photos albums returns limited set

skipmeister123, 2018-11-18:
When selecting an album to use during setup, I am presented with a limited subset of my actual albums.  Unfortunately, my target album isn't in this list.

skipmeister123, 2018-11-18:
Aha... read the code, and found https://github.com/laurentalacoque/TouchSelfie-extended/blob/master/scripts/setup.py#L278

laurentalacoque, 2018-11-18:
You're supposed to filter the list by typing Duke characters in the search field 😉

skipmeister123, 2018-11-18:
Yes, but it only filters the items pulled from the server. The amount of 
items being pulled was 15. Therefore, I could not find it when filtering.
--
Hans Kokx

laurentalacoque, 2018-11-18:
Hu oh!
This definitely is a bug 😉

laurentalacoque, 2018-11-22:
Hey @skipmeister123 !
I think that I do understand what's happening.

Actually, it seems that ALL albums are effectively pulled from your album list. 
See here:
https://github.com/laurentalacoque/TouchSelfie-extended/blob/fc4199da84a051b665494427efcca2e051add355/scripts/setup.py#L269

They are filtered inside a for loop that populates the list based on your filter.
BUT the 'limit to 15 items' condition is wrong here:
https://github.com/laurentalacoque/TouchSelfie-extended/blob/fc4199da84a051b665494427efcca2e051add355/scripts/setup.py#L295

because it's based on the original list index and not the current number of inserted elements.

This is addressed by https://github.com/laurentalacoque/TouchSelfie-extended/pull/57

@skipmeister123 could you please tell me if this fixes your issue?

skipmeister123, 2018-11-22:
I'm glad you figured it out. I actually solved the problem for myself by 
changing the value to something higher.  Since I've changed a lot of the 
other code, I wasn't going to pull in these changes to test.  I will bless 
it, in spirit, though. 😅
--
Hans Kokx

PR #57 (2019-01-18): inserted_items counting to fix bad filtering
This hopefully fixes #55
commit dac87963 (2019-01-18): inserted_items counting to fix bad filtering

This hopefully fixes #55
closed 2019-01-18 by laurentalacoque