laurentalacoque/TouchSelfie-extended #58: mmal errors - Failed to enable connection: Out of resources

andyhawkes, 2018-12-07:
I have just done a clean install of Rasbian (via NOOBS) and run through the setup process in the README, and everything is set up for the TouchSelfie app, but if I try to execute photobooth.sh I get the following error:

```
pi@photobooth:~ $ cd Desktop/photobooth/
pi@photobooth:~/Desktop/photobooth $ bash photobooth.sh 
mmal: mmal_vc_port_enable: failed to enable port vc.null_sink:in:0(OPQV): ENOSPC
mmal: mmal_port_enable: failed to enable connected port (vc.null_sink:in:0(OPQV))0xf6c070 (ENOSPC)
mmal: mmal_connection_enable: output port couldn't be enabled
Traceback (most recent call last):
  File "user_interface.py", line 942, in <module>
    ui = UserInterface(config,window_size=(SCREEN_W, SCREEN_H))
  File "user_interface.py", line 285, in __init__
    self.camera = mycamera.PiCamera()
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 433, in __init__
    self._init_preview()
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 513, in _init_preview
    self, self._camera.outputs[self.CAMERA_PREVIEW_PORT])
  File "/usr/lib/python2.7/dist-packages/picamera/renderers.py", line 558, in __init__
    self.renderer.inputs[0].connect(source).enable()
  File "/usr/lib/python2.7/dist-packages/picamera/mmalobj.py", line 2212, in enable
    prefix="Failed to enable connection")
  File "/usr/lib/python2.7/dist-packages/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to enable connection: Out of resources
```

I have checked that the camera is detected and supported via the following:

```
pi@photobooth:~ $ sudo vcgencmd get_camera
supported=1 detected=1
```

I have also increased GPU memory to 144

Nothing seems to make it want to work!

laurentalacoque, 2018-12-10:
Hey @andyhawkes !
I also ran into this kind of trouble.

This error takes place during the first instanciation of PiCamera()
This is frequently caused by : 
- System is not ready yet (e.g. you autolaunch the script at startup and the system is not ready yet)
- Another application is using the camera (e.g. a service for video surveillance)

This is not caused by TouchSelfie, any script that creates a PiCamera instance should fail the same way.

I advise you to:
- test if `raspistill` command line tool works (if it doesn't, there's an issue with your camera or there's another application using it)
- Search for solutions using the error as keywords (there are many reason why this doesn't work, none related to TouchSelfie)

Good luck!

closed 2018-12-10 by laurentalacoque

andyhawkes, 2018-12-10:
`raspistill` gives the same error.

I don't know what else could be using the camera though - it's literally a clean install of Raspbian with TouchSelfie set up on top of it.

I've ordered a new camera module and will test it with that once it arrives.

laurentalacoque, 2018-12-10:
@andyhawkes also double check if you don't have any background services that use the camera (e.g. 'motion' or camera/www bridge) and make sure that the camera is enabled with raspi-config

andyhawkes, 2018-12-30:
@laurentalacoque it looks like the v2 camera module I had was faulty - I swapped it out for a new v1 camera module and it works fine!