laurentalacoque/TouchSelfie-extended #59: Touche Selfie bloqued in a loop

karlzre, 2019-01-06:
Hi Laurent,

I had the issue 2-3 times, the touch Selfie start taking photos in a loop.

I think that happen if someone press a button to take a picture when it's busy with assembling the animated gif.

karlzre, 2019-01-16:
Hi Laurent, i just test it.

If you press on a button to take a picture during the "assembling animated gif" then touchselfie start takin a picture.

I think user started pushing all button because they didn't see the touche selfie reacting.

If you press on many button, then it will take all the pictures one after the others.

Would be maybe a good idea to show a bigger message "please wait" during image assembling.

laurentalacoque, 2019-01-18:
Hey @karlzre 
I'll try to reproduce this and fix it!

laurentalacoque, 2019-01-20:
Hey @karlzre !
What buttons are you using? Namely: is that a keyboard plugged to the raspberry or physical GPIO buttons?
I think that you're using a keyboard, can you confirm this?

laurentalacoque, 2019-01-20:
Hey @karlzre 
This should now be solved in `master`, just update your touchselfie ;)

karlzre, 2019-01-21:
I'm using a touchscreen, it's the button on the right.

laurentalacoque, 2019-01-22:
Ok, something still needs to be fixed then !

laurentalacoque, 2019-01-24:
Hi @karlzre 
This should now be solved in master

Thanks for your bug report!

closed 2019-01-24 by laurentalacoque