laurentalacoque/TouchSelfie-extended #60: Picasa API Deprecation Jan 2019

carolinedunn, 2019-01-17:
I am no longer able to upload photos to my Google Photos Album as Picasa API was depreciated on 1/15/2019. 
Please migrate to Google Photos Library API. Thank you!
Below text quoted from https://developers.google.com/picasa-web/

Picasa Web Albums Data API Deprecation
The Picasa Web Albums Data API will be turned off and all associated OAuth scopes will be disabled. Read the deprecation notice.

Key dates:

January 2019 Unless you have requested an extension, all calls to the Picasa Web Albums Data API will begin to fail from this date onwards.
March 2019 The Picasa Web Albums API will be turned off. All calls to this API will fail from this date onwards and you cannot request any further extensions.
Refer to the migration guide and start building with Google Photos APIs now.

laurentalacoque, 2019-01-18:
Will do!

laurentalacoque, 2019-01-20:
Update, this will require mucho trabajo :(
Moving this to high priority.
Any Help appreciated

karlzre, 2019-01-21:
Hi Laurent, 

I'm not a developper but if i can help in any way don't hesitate to ask. ( testing for examples)

laurentalacoque, 2019-01-28:
@PaulShadwell Everything is in file [`scripts/oauth2services.py`](https://github.com/laurentalacoque/TouchSelfie-extended/blob/master/scripts/oauth2services.py)

Methods that need to be changed:
 - `get_user_albums`
 - `upload_picture`
Alas, I'm not sure that the python google data API supports Google Photos API yet.
Also, the Google Photos API is a bit different from, now defunct, Picasa API because a photo updload requires two steps : actual uploading (provides an upload token) and file referencing (title, album association)

If you're willing to contribute, you just need to modify the file `oauth2services.py` and run it directly: it contains a `main` that tests message sending, photo uploading and album listing.

Again, any help is appreciated!

laurentalacoque, 2019-01-28:
# Food for thoughts

Here are some links to implement photos API in python:

 - [access albums (read only)](https://stackoverflow.com/a/54000367) 
 - [upload image and annotate it](https://stackoverflow.com/a/52038045)

laurentalacoque, 2019-01-28:
Created PR #61 to track the changes

PaulShadwell, 2019-01-28:
Thanks for the update Laurent. I assume from the pull request that you're working on it.
The google-api-python-client was last updated 3 days ago and is currently at v1.7.7 so looks like there shouldn't be an issue but I could be wrong.
Currently running some tests.

laurentalacoque, 2019-01-29:
Update: I successfully managed to download my album list today.

PaulShadwell, 2019-01-29:
Cool!! When can we expect the new build?

How much work would it be at add a choice of Photo services such as Flickr, Tumbler, Instagram, Facebook etc?

laurentalacoque, 2019-01-29:
@PaulShadwell no ETA yet, I don't use TouchSelfie anymore so I can only use spare time ;)
Maybe in a not so distant future, say... next week or two weeks from now?

As for the other services, you can post a new issue to ask for them, but I don't think I'll actively work on it because of a lack of motivation/time/knowledge of the APIs, but *please*, feel free to fork this repository, implement it and submit a pull request ;)

laurentalacoque, 2019-01-30:
Hey Guys, good news!

I think that the code in the [photoslibrary-api](https://github.com/laurentalacoque/TouchSelfie-extended/tree/photoslibrary-api) branch might be working!

To test it:
 - `git checkout photoslibrary-api` or download [from here](https://github.com/laurentalacoque/TouchSelfie-extended/tree/photoslibrary-api)
- remove `client_id.json`, `credentials.dat` files and go to the google dev console, add Photos Library API to your existing project and download the client secret to `scripts/google_client_id.json`
- launch `scripts/setup.py` again

**Note** : Photos API doesn't seem to allow uploading to albums currently, album selection was removed from setup

laurentalacoque, 2019-01-31:
Now complete with Setup.
Merged to master.
See important infos at the top of `Readme.md`

Don't hesitate to reopen if you get an issue

closed 2019-01-31 by laurentalacoque

PaulShadwell, 2019-01-31:
@laurentalacoque 
Great news Laurent. 
Thankyou for the great work.
Testing it out now.

karlzre, 2019-02-01:
@laurentalacoque <https://github.com/laurentalacoque>

Good job, i will try to find time this week end to test it.
As you know i've made changes in some files to manage a arduino in the
script, do i have to reconfigure after the update or is it only some files
to updates ?

laurentalacoque, 2019-02-01:
Hey @karlzre 
You can merge the differences into your codebase using `git pull` => this will add the new features but you might have to resolve conflicts manually.

Otherwise, you might just browse and implement manually the differences. [See this](https://github.com/laurentalacoque/TouchSelfie-extended/pull/61/files) for a list of changes

carolinedunn, 2019-02-01:
Thank you @laurentalacoque ! This is awesome. I did the git pull this morning and it works. In the process I created a new album called TouchSelfie. 
I am getting one small bug in the setup process. When I go to configure save files locally, I am getting a strange permission error. I close out of it, and everything seems to work just fine. Thank you again for your great work on this!

[image]
[image]

karlzre, 2019-02-02:
Hi Laurent,

i just did the git pull and it looks to work.

I had to rechange the personalised keyboard and add the lies to control de
Arduino.

if i undestand correctly it's no more (not yet) possible to create a new
album right ?

Thx

laurentalacoque, 2019-02-02:
@carolinedunn Hey ;)
Glad it works!

The IOError refers to the writing of the photobooth.sh script. Did you lock it or change its permission?
Maybe you created once as root and relaunch the setup as a simple user?

Anyway, there's no change in this file, if it was created once, then there's no problem reusing it :)

laurentalacoque, 2019-02-02:
@karlzre The good news is:
## You can now create and select albums again!

Points to keep in mind though:
 - I decided to as for the minimum of permissions
 - TouchSelfie asks for "appendonly" permission meaning you can append albums and photos
 - TouchSelfie limits photos and albums access **only to the items it created itself**

This is safer for everyone: this makes it impossible for TouchSelfie to accidentaly delete your own photos for example (or overwrite your albums).

This is better for privacy as well: TouchSelfie has no mean of dowloading or getting informations about your other photos

## How to create select or create a photo album?

Just use the "select album" button to browse the albums (only albums created by TouchSelfie will show up)
- The first time, just use the item `<create new>`, this will create an album "TouchSelfie" in your photostream with a random colored image (you can rename the album but don't remove the photo, because this will then delete the album)
- You can also select the `<No Album>` item: Photos will be uploaded to your photostream
- Once you have several albums created with the setup script, you can select the one you want this way too

Here's a snapshot of my albums after a day of coding/testing ;)
[image]

PaulShadwell, 2019-02-03:
Understood.
I was confused by the build instructions saying that the album had to be manually created first but this is much better.Thanks again.

commit 9e3e1a37 (2019-01-31): Implemented the new 'Google Photos Library API'

Move from Picasa Web API to Photoslibrary API: fixes #60