laurentalacoque/TouchSelfie-extended #62: Help wanted! Migrate to python 3.x

laurentalacoque, 2019-01-31:
Some of the frameworks on which TouchSelfie relies (such as google-api-client) are marking python 2.X as deprecated ([see this](https://github.com/googleapis/google-api-python-client#deprecated-python-versions)).
This might cause broken functionalities as soon as 2020.

If you want to give a hand, please fork, improve and submit PR!

Thanks :)

wyojustin, 2019-02-23:
Testing python3 branch now.  Everything seems to be working!
-- setup.py Google Photo Album entry is not writeable (have to stick with default "Drop Box")

wyojustin, 2019-02-23:
BTW, I copied the oath2services script into our directory and it worked first try!  Thanks!    Give it a try and test for me if you don't mind: https://github.com/wyolum/TouchSelfie/tree/python3

laurentalacoque, 2019-02-24:
> Testing python3 branch now. Everything seems to be working!
> -- setup.py Google Photo Album entry is not writeable (have to stick with default "Drop Box")

Thanks @wyojustin!
Could you please describe this error?
- is there any error in the console window?
- did you see/try the Modify button and "create directory" item?

**Note:** in the new google phtoos api, you **can't** see your existing google photos albums : you only see albums created with TouchSelfie

wyojustin, 2019-02-24:
Ok, now I get it.  Click Select to change the album.  I found this code on the web, and it CAN see all album names if you want to take a look.  https://gist.github.com/wyojustin/a41b7b98caee50ef1ede553e98480844

wyojustin, 2019-02-24:
I like the Long Press idea!  Cool implementation.  The first time I used it I thought the long press event would trigger automatically if I held long enought.  Of course, it only triggers when I RELEASE the press.  I modifed your code to trigger as soon as the long press duration has been reached using root.after.  See __press, __release, and __trigger in this gist:  https://gist.github.com/wyojustin/be3d2ac20020ebba8dbb8ae16442f694

laurentalacoque, 2019-02-25:
@wyojustin : thanks for the link about all albums recovery. Actually I already implemented it in a previous test version BUT, as a design choice, I preferred to limit the permissions to the strict minimum (append only).
This way, we can't accidentally mess with user's files 😉

laurentalacoque, 2019-02-25:
Also thanks for the longpress modification. Seems nice, I'll implement it ASAP

wyojustin, 2019-02-28:
Just found a bug in long press detector... gist updated.

laurentalacoque, 2019-02-28:
@wyojustin Nice !

wyojustin, 2019-03-09:
Laurent,  we added a annotation feature so you can write comments and thoughts.  Please test when you get a chance. https://github.com/wyolum/touchselfie

wyojustin, 2023-12-03:
Karlzre,
I'd try chainging "is" with "==" and see if that fixes the problem.  Good luck!
change

`f config.selected_printer is "": 
    config.selected_printer = selectedindex`

to 

`f config.selected_printer == "": 
    config.selected_printer = selectedindex`