laurentalacoque/TouchSelfie-extended #63: Getting invalid_scope

PaulShadwell, 2019-02-01:
When I run the setup.py (sudo python ./setup.py)

I checked my setup in Google and everything looks fine.
I'm sure I'm doing something wrong but not sure.

Any ideas?

laurentalacoque, 2019-02-01:
Hey @PaulShadwell 
I was able to reproduce the problem, I think I missed it because I already was identified.
Sorry!
The new version of master should fix this.
Can you confirm?

PaulShadwell, 2019-02-01:
Thanks @laurentalacoque
 I will test it again tonight.

PaulShadwell, 2019-02-01:
Hi @laurentalacoque, I was able to get it working but just one last issue. 
When I run setup.py I get the following issue:
1. Enable options page. Click Next
2. Google Account Green ticks. Click Next
3. Email Setup. Click Next
4. Google Photo Album. Click Select. Window opens I see <No Album> and <Create New> and that's it. None of my albums are displayed so I can't select any and consequence is no ID is collected.

How can I get to the settings without running setup.py?
I figured out the Long Touch to exit, I was unplugging the Pi till now to get out of the app. :-(

laurentalacoque, 2019-02-02:
Hey @PaulShadwell 
Glad this works finally!
Thanks for having pointed this :)

TouchSelfie asks for the minimum permissions of the Photos Library API: it can only *append* photos and browse albums that it has created *itself*.
This is why you don't see your albums in the album list: you must create one, then you'll see it.
This is done by selecting the "<Create new>" item in the "Select album": this will create a "TouchSelfie" album in your photostream were the photos will be uploaded. You can rename it and/or share it afterwards.

Hope this helps, see [the updated readme](https://github.com/laurentalacoque/TouchSelfie-extended/blob/master/README.md) for informations

closed 2019-02-02 by laurentalacoque