laurentalacoque/TouchSelfie-extended #66: Error : invalid scope

staff21, 2019-02-06:
Hello, no way to finish the ./setup.sh 
When i use the ./setup.sh a very helpfull help help me to create the google_
At the end, the raspi open a chronium with this link : 
 https://accounts.google.com/o/oauth2/auth?scope=https%3A%2F%2Fwww.googleapis.com%2Fauth%2Fphotoslibrary.appendonly+https%3A%2F%2Fwww.googleapis.com%2Fauth%2Fphotoslibrary.readonly.appcreateddatahttps%3A%2F%2Fwww.googleapis.com%2Fauth%2Fgmail.send&redirect_uri=http%3A%2F%2Flocalhost%3A8090%2F&response_type=code&client_id=1042781456115-4ctitr3vhvt3q5fbflo4lhi9ub4ov3se.apps.googleusercontent.com&access_type=offline

=> 400 that's an error.

staff21, 2019-02-07:
solved, i was not using the master source.

laurentalacoque, 2019-02-07:
Glad you found out ;)

closed 2019-02-07 by laurentalacoque