laurentalacoque/TouchSelfie-extended #67: Launch TouchSelfie at boot

laurentalacoque, 2019-02-17:
Although it's possible to launch TouchSelfie in rc.local or in the X server startup scripts, many services won't be ready and TouchSelfie will die on an unexpected error

For example, the piCamera module will fail until a few seconds _after_ X server has started.

We should therefore add a loop to try/except/sleep predictable errors and retry for a given period of time

laurentalacoque, 2019-02-17:
# note:
Simply `try`ing / `except` doesn't seem to work
The only solution so far : writing an `autostart.sh` script that does

```
#!/bin/sh
env sleep 10
/path/to/touchselfie/touchselfie.sh
```

Then add this line to `~/config/lxsessions/autostart`

```
@/path/to/autostart/autostart.sh
```

This way, TouchSelfie gets launched 10 seconds after Xserver has started (needed, otherwise picamera fails)

# todo
Write an autostart.sh script that launches `photobooth.sh` every 2 seconds until it works OR detects that Xserver has fully started

wyojustin, 2019-02-18:
Suggest you add TouchSelfie to main menu and launch from touchscreen.

laurentalacoque, 2019-02-23:
@wyojustin hey ;)

Already did :)
I'm looking for a fully resilient solution though.
Ideally, the photobooth should start on power-up without user action: this is more convenient whenever I lend it or rent it!

wearebeta, 2019-04-27:
I tried this solution but there is no "~/config/lxsessions/autostart" anymore?!

wyojustin, 2019-04-27:
If your pi won't boot, it's probably best to start with a clean install.

laurentalacoque, 2019-04-27:
@wearebeta
I think it's ‘.config‘ with a dot!

wearebeta, 2019-04-30:
I think, I found the solution:
https://gomilkyway.com/linux/how-to-run-a-command-at-the-start-of-raspbian-stretch/

"How to run a command at the start of raspbian stretch
Posted on February 22, 2019
To run a command at the start of raspbian stretch the newest raspbian on rasbperry 3+ for instance. you no longer use ~/.config/lxsession autostart. it is now in /etc/xdg/lxsession/LXDE-pi/autostart  because raspbian now uses P.I.X.E.L."

mattx38, 2020-04-09:
Otherwise, you can create a shortcut onto the desktop.
To do so, create a .desktop file and copy/paste this script (adapting the path to your personal configuration) :
`
[Desktop Entry]
Type=Application
Name=TouchSelfie
Comment=PhotoBooth
Exec=sudo /home/pi/TouchSelfie/photobooth.sh
Icon=face-smile-big
Terminal=false
Categories=Application
`