laurentalacoque/TouchSelfie-extended #69: Issue with running user_interface.py

ChristopherP1221, 2019-02-17:
Hello,

I think this project is awesome and you all have come a long way since the first version.  It appears I have the configuration working with Google and both the GMail and Photos APIs oauth scopes were added correctly.  However, when I try to run the final script, I get the following error:

pi@raspibooth:~/git/TouchSelfie-extended/scripts $ python user_interface.py
No handlers could be found for logger "__main__"
WARNING |10:37:13| install_key_binding: no action 'send_print'
WARNING |10:37:13| No hardware buttons found, generating software buttons
Traceback (most recent call last):
  File "user_interface.py", line 1028, in <module>
    ui = UserInterface(config,window_size=(SCREEN_W, SCREEN_H),log_level = logging.DEBUG)
  File "user_interface.py", line 310, in __init__
    self.camera.annotate_foreground = Color('white')
NameError: global name 'Color' is not defined

I don't have my touchscreen attached yet.  I'm simply trying to get things working on a regular monitor.  I have confirmed my picamera can take still photos and videos.  When I run the script to start UI, it appears to load for a moment and then returns the error above.  I only see a gray screen for about 1 second.  Any help or guidance you can give would be very much appreciated.

Hardware:
Raspberry Pi 3 B+
PiCam v2
Touchless monitor hooked up through an HDMI cable

laurentalacoque, 2019-02-17:
Hey @ChristopherP1221 !
Sorry to hear that.

Could you please `git pull` to get the last version and paste the log again?

I think that maybe something went south with `picamera` import

ChristopherP1221, 2019-02-17:
Looking better!  I'm able to get a UI now but nothing from the camera.  I wonder what I did to break the original pull... oh well.  Anyway, stack trace below, maybe I'm still missing packages?  Before I reached out, I was getting an error for cv2 as well if that helps you at all.  As usual, any guidance is appreciated!

pi@raspibooth:~/git/TouchSelfie-extended $ sh photobooth.sh 
WARNING:__main__:picamera not found, trying cv2_camera
DEBUG:user_interface:Installing keybinding 's' for action 'snap_None'
DEBUG:user_interface:Installing keybinding 'S' for action 'snap_None'
DEBUG:user_interface:Installing keybinding '<F1>' for action 'snap_None'
DEBUG:user_interface:Installing keybinding 'f' for action 'snap_Four'
DEBUG:user_interface:Installing keybinding 'F' for action 'snap_Four'
DEBUG:user_interface:Installing keybinding '<F2>' for action 'snap_Four'
DEBUG:user_interface:Installing keybinding 'a' for action 'snap_Animation'
DEBUG:user_interface:Installing keybinding 'A' for action 'snap_Animation'
DEBUG:user_interface:Installing keybinding '<F3>' for action 'snap_Animation'
DEBUG:user_interface:Installing keybinding 'e' for action 'send_email'
DEBUG:user_interface:Installing keybinding '@' for action 'send_email'
DEBUG:user_interface:Installing keybinding '<Escape>' for action 'configure'
WARNING:user_interface:install_key_binding: no action 'send_print'
WARNING |17:51:04| install_key_binding: no action 'send_print'
WARNING:user_interface:No hardware buttons found, generating software buttons
WARNING |17:51:04| No hardware buttons found, generating software buttons
DEBUG:oauth2services:refresh: Refreshing authentication token
DEBUG:oauth2services:__oauth_login: getting cached authentication token

ChristopherP1221, 2019-02-18:
Success!  I installed the picamera module using pip and things seem to be working now. I will tinker with this a bit and circle back if I run into any problems or have any questions.  Thanks for the initial assist, I will circle back if I add anything cool!

closed 2019-02-18 by ChristopherP1221