laurentalacoque/TouchSelfie-extended #73: Error uploading

karlzre, 2019-03-03:
Hi Laurent,

Singe today i have a message "error uploading"

What can be the cause of this message ?

I ve  check the pi has acces to internet.

laurentalacoque, 2019-03-06:
Hey @karlzre
Sorry to hear that 

Could you paste the /scripts/photobooth.log here?

karlzre, 2019-03-06:
hi laurent

here are the logs.
now i ve tempoay diabled the upload

2019-03-05 19:57:56|user_interface  | DEBUG   | Installing keybinding 's' for action 'snap_None'
2019-03-05 19:57:56|user_interface  | DEBUG   | Installing keybinding 'S' for action 'snap_None'
2019-03-05 19:57:56|user_interface  | DEBUG   | Installing keybinding '<F1>' for action 'snap_None'
2019-03-05 19:57:56|user_interface  | DEBUG   | Installing keybinding 'f' for action 'snap_Four'
2019-03-05 19:57:56|user_interface  | DEBUG   | Installing keybinding 'F' for action 'snap_Four'
2019-03-05 19:57:56|user_interface  | DEBUG   | Installing keybinding '<F2>' for action 'snap_Four'
2019-03-05 19:57:56|user_interface  | DEBUG   | Installing keybinding 'a' for action 'snap_Animation'
2019-03-05 19:57:56|user_interface  | DEBUG   | Installing keybinding 'A' for action 'snap_Animation'
2019-03-05 19:57:56|user_interface  | DEBUG   | Installing keybinding '<F3>' for action 'snap_Animation'
2019-03-05 19:57:56|user_interface  | DEBUG   | Installing keybinding 'e' for action 'send_email'
2019-03-05 19:57:56|user_interface  | DEBUG   | Installing keybinding '@' for action 'send_email'
2019-03-05 19:57:56|user_interface  | DEBUG   | Installing keybinding '<Escape>' for action 'configure'
2019-03-05 19:57:56|user_interface  | WARNING | install_key_binding: no action 'send_print'
2019-03-05 19:57:56|tkImageLabel    | DEBUG   | Created ImageLabel with size (1024, 600)
2019-03-05 19:57:57|user_interface  | WARNING | No hardware buttons found, generating software buttons
2019-03-05 19:57:58|oauth2services  | DEBUG   | refresh: Refreshing authentication token
2019-03-05 19:57:58|oauth2services  | DEBUG   | __oauth_login: getting cached authentication token
2019-03-05 19:58:00|user_interface  | INFO    | Snaping photo (mode=None)
2019-03-05 19:58:00|tkImageLabel    | DEBUG   | Unloading image
2019-03-05 19:58:00|user_interface  | DEBUG   | snap: single picture
2019-03-05 19:58:06|user_interface  | DEBUG   | snap: adding logo
2019-03-05 19:58:07|user_interface  | DEBUG   | snap: saving snapshot
2019-03-05 19:58:07|user_interface  | DEBUG   | snap: displaying image
2019-03-05 19:58:07|tkImageLabel    | DEBUG   | Loading image 'snapshot.jpg'
2019-03-05 19:58:08|user_interface  | DEBUG   | Uploading image
2019-03-05 19:58:08|user_interface  | INFO    | Image 05-03-2019 19:58:07 successfully uploaded
2019-03-05 19:58:08|user_interface  | DEBUG   | Archiving image 05-03-2019 19:58:07
2019-03-05 20:04:13|user_interface  | INFO    | Snaping photo (mode=Four)
2019-03-05 20:04:13|tkImageLabel    | DEBUG   | Unloading image
2019-03-05 20:04:13|user_interface  | DEBUG   | snap: starting collage of four
2019-03-05 20:04:30|user_interface  | DEBUG   | snap: assembling collage
2019-03-05 20:04:30|user_interface  | DEBUG   | snap: Adding  the collage cover
2019-03-05 20:04:31|user_interface  | DEBUG   | snap: Saving collage
2019-03-05 20:04:32|user_interface  | DEBUG   | snap: displaying image
2019-03-05 20:04:32|tkImageLabel    | DEBUG   | Loading image 'collage.jpg'
2019-03-05 20:04:32|user_interface  | DEBUG   | Uploading image
2019-03-05 20:04:32|user_interface  | INFO    | Image 05-03-2019 20:04:32 successfully uploaded
2019-03-05 20:04:32|user_interface  | DEBUG   | Archiving image 05-03-2019 20:04:32
2019-03-05 20:05:24|user_interface  | INFO    | Snaping photo (mode=None)
2019-03-05 20:05:24|tkImageLabel    | DEBUG   | Unloading image
2019-03-05 20:05:25|user_interface  | DEBUG   | snap: single picture
2019-03-05 20:05:31|user_interface  | DEBUG   | snap: adding logo
2019-03-05 20:05:31|user_interface  | DEBUG   | snap: saving snapshot
2019-03-05 20:05:31|user_interface  | DEBUG   | snap: displaying image
2019-03-05 20:05:31|tkImageLabel    | DEBUG   | Loading image 'snapshot.jpg'
2019-03-05 20:05:32|user_interface  | DEBUG   | Uploading image
2019-03-05 20:05:32|user_interface  | INFO    | Image 05-03-2019 20:05:31 successfully uploaded
2019-03-05 20:05:32|user_interface  | DEBUG   | Archiving image 05-03-2019 20:05:31
2019-03-05 20:12:41|user_interface  | INFO    | Snaping photo (mode=None)
2019-03-05 20:12:41|tkImageLabel    | DEBUG   | Unloading image
2019-03-05 20:12:41|user_interface  | DEBUG   | snap: single picture
2019-03-05 20:12:47|user_interface  | DEBUG   | snap: adding logo
2019-03-05 20:12:48|user_interface  | DEBUG   | snap: saving snapshot
2019-03-05 20:12:48|user_interface  | DEBUG   | snap: displaying image
2019-03-05 20:12:48|tkImageLabel    | DEBUG   | Loading image 'snapshot.jpg'
2019-03-05 20:12:48|user_interface  | DEBUG   | Uploading image
2019-03-05 20:12:48|user_interface  | INFO    | Image 05-03-2019 20:12:48 successfully uploaded
2019-03-05 20:12:48|user_interface  | DEBUG   | Archiving image 05-03-2019 20:12:48
2019-03-05 20:12:50|user_interface  | INFO    | Snaping photo (mode=Four)
2019-03-05 20:12:50|tkImageLabel    | DEBUG   | Unloading image
2019-03-05 20:12:50|user_interface  | DEBUG   | snap: starting collage of four
2019-03-05 20:13:12|user_interface  | INFO    | Effect sketch selected
2019-03-05 20:13:14|user_interface  | INFO    | Effect sketch selected
2019-03-05 20:13:15|user_interface  | INFO    | Effect sketch selected
2019-03-05 20:13:15|user_interface  | DEBUG   | snap: assembling collage
[... 1014 lines cut ...]
2019-03-06 16:40:27|user_interface  | DEBUG   | Uploading image
2019-03-06 16:40:27|user_interface  | INFO    | Image 06-03-2019 16:40:26 successfully uploaded
[... cut ...]
2019-03-06 16:40:40|user_interface  | DEBUG   | snap:animation: capturing image 10
2019-03-06 16:40:40|user_interface  | DEBUG   | snap: assembling animation
2019-03-06 16:43:59|user_interface  | INFO    | Effect negative selected
2019-03-06 19:08:40|user_interface  | INFO    | Effect negative selected
2019-03-06 19:08:46|user_interface  | INFO    | Effect sketch selected
2019-03-06 19:08:50|user_interface  | INFO    | Effect gpen selected
2019-03-06 19:09:12|user_interface  | INFO    | Effect colorswap0 selected

karlzre, 2019-03-06:
Laurent,

I m actually in France in a chalet with 40 child playing with touchselfy
they like it +-100 pictures a day :)

I think the Issue may be the bad wifi connection.

Sometimes  i also found the touch selfy bloqued.
Nothing happend when I touch  a button I only see the mouse arrow on the
screen .

Problem is I have no idea what the child's do with the touchselfy  .

laurentalacoque, 2019-03-06:
Hi @karlzre 

Yes, I think that's just intermitent deconnections (bad Wifi or wifi down)

Error reports a problem in the low level internet access : `ServerNotFound` which means that there's a DNS or Network error. This isn't related to TouchSelfie ;)

Here's an example:

```
2019-03-06 16:17:05|user_interface | ERROR | send_picture: Mail sending Failed
Traceback (most recent call last):
File "user_interface.py", line 884, in __send_picture
self.last_picture_filename)
File "/home/pi/TouchSelfie-extended/scripts/oauth2services.py", line 288, in send_message
credentials = self.__oauth_login()
File "/home/pi/TouchSelfie-extended/scripts/oauth2services.py", line 101, in __oauth_login
credentials.refresh(Http())
File "/usr/local/lib/python2.7/dist-packages/oauth2client/client.py", line 545, in refresh
self._refresh(http)
File "/usr/local/lib/python2.7/dist-packages/oauth2client/client.py", line 761, in _refresh
self._do_refresh_request(http)
File "/usr/local/lib/python2.7/dist-packages/oauth2client/client.py", line 780, in _do_refresh_request
body=body, headers=headers)
File "/usr/local/lib/python2.7/dist-packages/oauth2client/transport.py", line 282, in request
connection_type=connection_type)
File "/usr/local/lib/python2.7/dist-packages/httplib2/init.py", line 2133, in request
cachekey,
File "/usr/local/lib/python2.7/dist-packages/httplib2/init.py", line 1796, in _request
conn, request_uri, method, body, headers
File "/usr/local/lib/python2.7/dist-packages/httplib2/init.py", line 1707, in _conn_request
raise ServerNotFoundError("Unable to find the server at %s" % conn.host)
ServerNotFoundError: Unable to find the server at oauth2.googleapis.com
2019-03-06 16:17:05|user_interface | ERROR | Error sending email
```

closed 2019-03-06 by laurentalacoque