laurentalacoque/TouchSelfie-extended #74: customizations

wearebeta, 2019-04-01:
Hello, 
I want to make some adjustments:
1. display an image with the hint to touch the button/screen to start
2. four photos will be taken with Countdown (as it is currently)
3. after the finished collage is displayed, the print button should also be displayed

I was able to hide the following function thanks to the "constants.py":
- Send via e-mail
- animation

Since I have little to no idea of Phyton, I would be happy about any help. Gladly also for some remuneration.

tiffanieanh, 2019-06-08:
Can I ask how you hid the animation button? It let me hide the effects pannel but for some reason when I try to adjust the code to hide the animations button it gives me an error pertaining to user_interface.py because of the constants import I think.

kyle-jennings, 2019-07-08:
@tiffanieanh remove the button from SOFTWARE_BUTTONS in constants.py

closed 2019-11-05 by laurentalacoque

mainstreet88, 2019-12-04:
Can someone explain in more detail how to "remove the button from SOFTWARE_BUTTONS in constants.py" . I have tried deleting different lines and it always crashes. Please help me.

Froopster, 2020-08-03:
To remove the animation button open the constants.py file.  Delete the comma in line 70 and everything in line 71 through 73.  Keep the bracket on line 74.  Hope this helps in the future.