laurentalacoque/TouchSelfie-extended #79: Select other templates

djtale, 2019-06-24:
Hello !

Thanks a lot for this script.
I have modified it a bit to be able to change the watermark on the fly but I would like to be able to select different templates for the collage_four_square.
Let say I have multiple 4_square layouts and I would like to select one.
Would it be possible ?

I actually have some difficulties to update the user_interface.py code and this section :
                #snapshot.paste(Image.open('collage_1.jpg'), (  0,   0,  w, h))
                #snapshot.paste(Image.open('collage_2.jpg'), (w,   0, w_, h))
                #snapshot.paste(Image.open('collage_3.jpg'), (  0, h,  w, h_))
                #snapshot.paste(Image.open('collage_4.jpg'), (w, h, w_, h_))

Would it be possible to have some explanation and advices if I want to change the code to fit a specific layout ? How to find out the size and position of the snapshots...

Thanks a lot.

laurentalacoque, 2019-07-09:
Hey @djtale 

In collage mode, the camera takes 4 pictures named "collage_X.jpg" with X ranging from 1 to 4.

In the lines you highlighted here, the four images are combined into a new one named "snapshot".
The line `snapshot.paste( im1, 0, 0, w, h)` indicates to paste image `im1` into the image `snapshot` at location `(x,y) = (0,0)` and `width/height = w,h` (see the [PIL reference for paste](https://pillow.readthedocs.io/en/stable/reference/Image.html#PIL.Image.Image.paste) )

To summarize:

if you want to change the relative arrangement of snapshots, you should modify these lines to 1/ create a blank image of the desired final resolution and 2/ paste the snapshots where you want them.

Hope I'm clear enough ;)

djtale, 2019-08-02:
Hello Laurent,

thanks a lot for the explanation.
I managed to do those changes among others:
- customize watermark with keyboard display in the UI
- select a template with a dropdown menu
- menu to restart/kill app, restart/shutdown system.

For the moment I'm using a PiCamera. Do you know if this script can work with a real camera such as a Canon EOS 77D connected with USB to the RapsberryPi ?

Thanks.

laurentalacoque, 2019-11-05:
Hello :)
This script was done for PiCamera so no, not easily ;(

For any other camera (webcams / DSLR), one should:
 - make image capture on the raspberry pi (without touchselfie)
 - implement a module (e.g. `scripts/my_camera.py`) taking example of `scripts/cv2_camera.py`
 - edit the `Camera::capture()` method in order to take the picture

closed 2019-11-05 by laurentalacoque