laurentalacoque/TouchSelfie-extended #81: How to flip the image taken ?

djtale, 2019-08-06:
Hello,

I have found in user_interface the code to flip the image of the camera.
But is it possible to flip the image that is saved and displayed on the screen ?

Thanks.

laurentalacoque, 2019-08-07:
Hey djtale :)

If you want both the preview and the snapshots to be flipped, you just have to set these:

```py
        self.camera.preview.hflip = True  #Mirror effect on preview for easier selfies
        self.camera.hflip = True  #Mirror effect on captured images
```
The photobooth code currently only flips the preview (so that the preview looks like a mirror) but doesn't flip the output (so that text in the picture isn't written right to left)

For more control over your captured images, you can also use the following:
To flip captured images you can use a [geometrical transform from PIL library](https://pillow.readthedocs.io/en/stable/handbook/tutorial.html?highlight=flip#geometrical-transforms)

```py
snapshot = Image.open('snapshot.jpg') # open the snapshot
snapshot = snapshot.transpose(Image.FLIP_LEFT_RIGHT) # flip it
snapshot.save("snapshot.jpg") # save it
```

closed 2019-08-07 by laurentalacoque

djtale, 2019-08-07:
Thanks a lot Laurent.

Indeed I have to take care of the watermark I apply to the image.

I'll do some tests.