laurentalacoque/TouchSelfie-extended #82: Canon Selphy CP 1300 not printin

hansamann, 2019-08-27:
Hi all, 

I have cups installed and the CP 1300 added as a printer, with the CP 900 driver (I tried the 1300 driver before) with gutenprint. When trying to print, I get a printing error that says 

Failure to send data to printer (libusb error -1 (2560/131072 to 0x01))

Any ideas? the printer is of course added and connected via usb to the raspberry pi.

hansamann, 2019-08-28:
> Sorry, no idea :(
>
> Did you manage to print something outside of touchselfie?

Fixed, will update the issue once I find time. I used the 910 driver (not
900, not 1300) and that works...

Thx, great Project!

Sven

laurentalacoque, 2019-08-28:
Sorry, no idea :(

Did you manage to print something outside of touchselfie?

closed 2019-08-28 by laurentalacoque