laurentalacoque/TouchSelfie-extended #83: Canon Selphy 1300 printing issues, again

hansamann, 2019-09-02:
Hi Laurent, all, 

I am using the Canon SELPHY CP910 - CUPS+Gutenprint v5.3.1 printing driver withthe CP1300 printer - we recently closed an issue after it would not print with the CP1300 driver... I switched to the CP910 driver and it worked a few prints. 

Now the issue is sadly back, I also must assume it never worked consistently... 

I opened up CUPS a bit now to allow me to browse the CUPS web interface from my laptop. the exact error I still get is this:

Idle - "Failure to send data to printer (libusb error -1: (25600/262144 to 0x01))"

As I just sat here and write the issue, the printer now suddenly started printing. So it seems to work from time to time, but it does not consistently get the jobs submitted. The printer interface shows "busy, receiving data" each time I click the printer. But then often it does nothing, as it seems it cannot transfer correctly?

Bad USB cable? Mmmh. Any ideas?

laurentalacoque, 2019-09-03:
Hello @hansamann :)

As noted in the other thread, this doesn't seem to be a TouchSelfie issue.
This is the part in TouchSelfie that is responsible for printing a picture (in `user_interface.py`)

```python

def send_print(self):
        self.log.debug("send_print: Printing image")
        try:
            conn = cups.Connection()
            printers = conn.getPrinters()
            default_printer = printers.keys()[self.selected_printer]#defaults to the first printer installed
            cups.setUser(getpass.getuser())
            conn.printFile(default_printer, self.last_picture_filename, self.last_picture_title, {'fit-to-page':'True'})
            self.log.info('send_print: Sending to printer...')
        except:
            self.log.exception('print failed')
            self.status("Print failed :(")
        self.log.info("send_print: Image printed")
```
You might want to check the logfile: it should read "send_print: Image printed"

Any other issue is linked to the [`pycups`](https://github.com/OpenPrinting/pycups) python module or your cups installation. You might want to ask your question there for better support?

Maybe start from the start and try to print an image using command line tool `lp` and see if this works? [Here's some command lines](https://www.cups.org/doc/options.html) you can use with CUPS

hansamann, 2019-09-07:
Thx a lot @laurentalacoque - I now got a new usb cable and the printing is unfortunately still a bit flaky. Is there an easy way to deactivate the printing, e.g. go with uploading to the photo album only?

hansamann, 2019-09-07:
Answering my own - there's the configuration.json where it can be turned off. I also had issues with the apsect ratio - I think the pi cam is 16:9 but the printer is 3:2, which strangely stretched my images on second look. I am now going with uploading the images and then printing from my phone. At least for me, the cups queue got stuck all the time. 

Thx for all the help, I think I found the best solution for me!

closed 2019-11-05 by laurentalacoque