laurentalacoque/TouchSelfie-extended #85: Preview mode : always

lliwgo, 2019-11-04:
Dear, it seems that there is no attract mode to attract peoples.
It is possible to have an attract mode with a preview ?

what is the command line (ssh) to exectute a picture at distance ??

laurentalacoque, 2019-11-05:
Hello @lliwgo 

Both should be possible with a little work.
I'm currently out of time, but here's how it could be done : 

# Attract mode

The attract mode could be a gif animation produced externally (gimp/photoshop)
This gif image would be loaded in the display.
Loading an image or a gif in the display is achieved in `scripts/user_interface.py` in any method of the class `UserInterface` using the command `self.image.load("attract_anim.gif")`

To launch the attract animation at startup, one could simply add `self.image.load("attract_anim.gif")` in the class initializations, for example around line 192:

```py
        #Configure Image holder (scripts/user_interface.py:192)
        self.image = ImageLabel(self.root, size=self.size)
        self.image.place(x=0, y=0, relwidth = 1, relheight=1)
        self.image.configure(background='black')
        # load attract animation (add this)
        self.image.load("attract_anim.gif")
```

Attract mode could be rescheduled a few minutes after each snapshot in the `User_Interface::snap` method using `Tk:after` calls.

One should define a `User_Inteface::launch_attract` method containing the same call as previously. and a `User_Interface::attract_after_id` alarm holder, then in `User_Interface::snap` method, it would be possible to do the following:

```py
def snap(self,mode="None"):
        # All
        # Existing
        # Code
        # Until
        # Final
        # Return

        # cancel any previously launched alarm
        self.root.after_cancel(self.attract_after_id)
        # program the attract mode in a minute
        self.root.after(60*1000, self.launch_attract)
        
        return snap_filename

```

# ssh Snaps

Sadly, there isn't any easy way to connect the system ssh to the event based Tk loop.

Some thoughts though:
 - create a polling mechanism (see `User_Interface::run_periodically` method)
 - make this polling method watch for the presence of a file (e.g. `scripts/ssh_snap_request`)
 - if the file is present, 
    - delete it
    - call the method `self.snap()`

Now using ssh, you can simply execute `touch /path/to/touchselfie/scripts/ssh_snap_request`

djtale, 2019-12-31:
Hi @laurentalacoque ,

The attract_mode is working fine when I launch the application the first time.
But I don't understand how to setup the alarm holder to run the attact_mode after a while.

I have defined that method :
    def launch_attract(self):
        self.image = ImageLabel(self.root, size=self.size)
        self.image.place(x=0, y=0, relwidth = 1, relheight=1)
        self.image.configure(background='black')
        # load attract animation (add this)
        self.image.load("attract_anim.gif")

>> Here I'm not sure if I'm supposed to have the whole block or just the last line.

And where should I define self.attract_after_id and what value should it be set to ?

Thanks

djtale, 2020-01-20:
Hi Laurent,

Could you please clarify how you define the alarm holder :
User_Interface::attract_after_id alarm holder

Thanks

Froopster, 2020-09-16:
Did you ever get this to work?  I'd like to do the same thing.
Thanks.

djtale, 2020-09-17:
Hi Froopster,

No I forgot about that feature.
I will give it a try later.

If I manage to make it working, I'll keep you updated