laurentalacoque/TouchSelfie-extended #86: snap portion (main photo) not working, printer not working

duamatef, 2019-11-14:
laurent, 
you have created and excellent product but I have problem.  I am a raspberry pi newbie running touchselfie-extended on a raspberry pi 3 B+.  I am using the official raspberry pi screen 7" and a raspberry pi camera v1. The program works fine except for two problems which may or may not be related.  The collage photos and the GIF image work and can be emailed, but the the "snap" photo cannot.  In addition none of the three can be printed.  I am attaching the error code for the inability to make a snap photo and the error code for the inability to print.  I can only send them as screen shots.  I apologize.  I have a suspicion that the v1 camera is the culprit based on the error message, but in the constants.py file the program seems to accomodate the v1 camera.  Please help.  Thank you so much for your great product.
[image]
[image]

duamatef, 2019-11-14:
sorry the pics are correctly oriented on my computer.  Not sure why they rotated on upload.  apologies.

laurentalacoque, 2019-11-14:
hello @duamatef :)

# Fix resolution errors when using pi camera v1

Good news,: TouchSelfie is easily adaptable to V1 Camera.

Edit the file `scripts/constants.py` and scroll to the resolution option:

```py

# Parameters for the three main effects
# None: simple shot
# Four: Collage of four shots
# Animation : animated gif
v2_full_size = (3280,2464)
v2_half_size = (1640,1232)
v2_quarter_size = (820,616)

v1_full_size = (2592,1944)
v1_half_size = (1296,972)
v1_quarter_size = (648,486)

EFFECTS_PARAMETERS = {
    "None": {
        'snap_size' : v2_full_size, #(width, height) => preferably use integer division of camera resolution
        'logo_size' : 128,         # height in pixels of the logo (will be thumbnailed to this size)
        'logo_padding' : 32        # bottom and right padding of the logo (pixels)
    },
    "Four": { 
        'snap_size' : v2_half_size,                       #(width, height) of each shots of the 2x2 collage
        'foreground_image' : "collage_four_square.png" # Overlay image on top of the collage
    },
    "Animation": {
        'snap_size' : (500, 500),   #(width, height) => Caution, gif animation can be huge, keep this small
        'frame_number' : 10,        # number of frames in the animation
        'snap_period_millis' : 200, # time interval between two snapshots
        'gif_period_millis' : 50    # time interval in the animated gif
    }
}

```

As you can see, the parameters `"snap_size'` control the resolution of the shots. You can safely change any `v2_XXX_size` to `v1_XXX_size`  in the `EFFECTS_PARAMETERS` `dict`.
In your case this would translate to:

```py

# Parameters for the three main effects
# None: simple shot
# Four: Collage of four shots
# Animation : animated gif
v2_full_size = (3280,2464)
v2_half_size = (1640,1232)
v2_quarter_size = (820,616)

v1_full_size = (2592,1944)
v1_half_size = (1296,972)
v1_quarter_size = (648,486)

EFFECTS_PARAMETERS = {
    "None": {
        'snap_size' : v1_full_size, #CHANGED (width, height) => preferably use integer division of camera resolution
        'logo_size' : 128,         # height in pixels of the logo (will be thumbnailed to this size)
        'logo_padding' : 32        # bottom and right padding of the logo (pixels)
    },
    "Four": { 
        'snap_size' : v1_half_size,   #CHANGED (width, height) of each shots of the 2x2 collage
        'foreground_image' : "collage_four_square.png" # Overlay image on top of the collage
    },
    "Animation": {
        'snap_size' : (500, 500),   #(width, height) => Caution, gif animation can be huge, keep this small
        'frame_number' : 10,        # number of frames in the animation
        'snap_period_millis' : 200, # time interval between two snapshots
        'gif_period_millis' : 50    # time interval in the animated gif
    }
}

```

# Solve printing issues

Sorry, I won't be able to provide support for this one.
This feature was contributed by someone else and I don't have any printers to debug it :(
Some rules of thumb though:
- Please ensure that you can print from your raspberry to your printer OUTSIDE of TouchSelfie, if not, please try to find some support in the cups forum
- Try running the setup again in order to reconfigure your printer

duamatef, 2019-11-16:
Laurent,
Thanks so much.  That worked!  The "snap" portion now works.  I will continue to work on the printer.  Can you also help me with the customization of the configuration.json file.  Below is a copy of my file:

{
    "archive_to_all_usb_drives": true, 
    "countdown_before_snap": 5, 
    "countdown_inter_snap": 3, 
    "email_body": "Greetings, here's your photo sent from the photobooth\n\n\n", 
    "email_subject": "Here's your photo!", 
    "enable_effects": true, 
    "enable_email": true, 
    "enable_email_logging": true, 
    "enable_hardware_buttons": false, 
    "enable_print": true, 
    "enable_upload": true, 
    "full_screen": true, 
    "gmail_user": "XXXXXphotobooth@gmail.com", 
    "google_photo_album_id": "None", 
    "google_photo_album_name": "Drop Box", 
    "local_archive": true, 
    "local_archive_dir": "/home/pi/TouchSelfie/Photos", 
    "logo_file": "capgi4.png", 
    "selected_printer": 1, 
    "snap_caption": "MERRY XMAS"
}

Based on the above I would expect the "MERRY XMAS" to be displayed and the logo file which is 1500 x 600 to displayed as well.  

[image]

The snap photo shows that the "MERRY XMAS" is not printed and the logo is super tiny.  I tried increasing the dimensions of the logo but it is still super tiny.  Please help and thank you in advance

laurentalacoque, 2020-02-08:
@duamatef 
Sorry for the late answer, you surely have solved your problem by now.

A few thoughts though:
- The caption configuration key is used to set the caption on the photo album, not on the photo itself
- The logo size and position is governed by the `EFFECTS_PARAMETERS` configuration dict in file `scripts/constants.py`:

```py
EFFECTS_PARAMETERS = {
    "None": {
        'snap_size' : v2_full_size, #(width, height) => preferably use integer division of camera resolution
>        'logo_size' : 128,         # height in pixels of the logo (will be thumbnailed to this size)
>        'logo_padding' : 32        # bottom and right padding of the logo (pixels)
    },
    "Four": { 
        'snap_size' : v2_half_size,                       #(width, height) of each shots of the 2x2 collage
        'foreground_image' : "collage_four_square.png" # Overlay image on top of the collage
    },
    "Animation": {
        'snap_size' : (500, 500),   #(width, height) => Caution, gif animation can be huge, keep this small
        'frame_number' : 10,        # number of frames in the animation
        'snap_period_millis' : 200, # time interval between two snapshots
        'gif_period_millis' : 50    # time interval in the animated gif
    }
}
```

closed 2020-02-08 by laurentalacoque