laurentalacoque/TouchSelfie-extended #88: Hardware Buttons not recognized?

Anekd0t, 2020-02-07:
Hi!
Thank you for a great project.

However I can for the love of my life not get the hardware buttons to work.
I am using this button from adafruit: [https://learn.adafruit.com/press-your-button-for-raspberry-pi/assembly] with the same suggested wiring.

I have tried putting the wire on both 8, 10 and 12 pins. I have also tried changing them to 11, 13 and 15 in without any difference. Tried different wires and different switches. I have tried both pull_up and pull_down and changed active state in constants.py.
The led on the  button is working but upon quitting photo booth, I am presented with: “No hardware buttons found, generating software buttons” every time.

Anyone have some input on what I am doing wrong?
Cheers

laurentalacoque, 2020-02-08:
Hey @Anekd0t 😉
Are you sure that you installed the gpio python module?
You can check this by laughing the python interpreter and importing the module. 

```
$ python
>>>     import RPi.GPIO as GPIO
```
If this raises an exception, then you don't have it and should install it.
```
sudo apt-get install python-rpi.gpio
```
Should do the trick

Anekd0t, 2020-02-08:
Hey @laurentalacoque !

Thank you for the quick reply!

Ok, so I ran “python”, then “import RPi.GPIO as GPIO” in the terminal, and I was not presented with any errors, so I figure it was already imported?

If I run “sudo apt-get install python-rpi.gpio” it said it was already installed.

Still getting “No hardware buttons found” error..

Question: I am also only using 1 button as of now if that matters?
Question: I can change pins while photobooth.sh is running right, since it’s polling for hardware buttons every 100ms it should remove the software button where responding pin is connected?

Cheers

laurentalacoque, 2020-02-08:
Hey @Anekd0t 
Ok, simplest thing to do is to debug `Buttons` interface independently.
Luckily enough, you can do this by launching the module itself.
For this:
```
$ cd scripts
$ python hardware_buttons.py
```
This script reports button presses and releases.

You might want to change the preassigned buttons pins, for this change line 105 from
```py
    buttons = Buttons()
```
to
```py
    buttons = Buttons(button_pins=[12,13]) # change with your actual pins numbers
    # note: it's ok to have a long list of pins, put any pin you want or even list(range(25)) to be sure
```
Then restart `hardware_button.py` and check if the key presses reports any event.

Anekd0t, 2020-02-08:
Hey again @laurentalacoque ,

Ok, so I’m not sure if I’m following your directions accurately, but I ran hardware_button.py in the terminal before and after the change of line 105. I changed the pins to 10,8 and 12. Button is connected to pin 8.
This is what I got:
[image]

Cheers

laurentalacoque, 2020-02-08:
it's `buttons = Buttons(buttons_pins=[12,13])` (mind the final s at buttons in buttons_pins) 
My bad.

Anekd0t, 2020-02-09:
Hey @laurentalacoque ,

Sorry, added the s.
Still similar error:
[image]

If I run it in the python editor(?) I get this as response:
[image]

Cheers for helping out! :)

Anekd0t, 2020-02-17:
Hey @laurentalacoque 
Had any time to look at the previous comment?

Cheers

Anekd0t, 2020-02-18:
Hey @laurentalacoque 
I just wanna say I figured it out, it seems the script is not recognizing my buttons since they don’t send a signal to the pins unless grounded.
That’s probably why the script says “no hardware buttons found”.
If I hold down the buttons while starting photobooth, the software buttons are removed.
So I went ahead and commented out the “if no hardware buttons” in user_interface.py, since I know I have buttons configured - then changed active state to 0. 
Now it all works as intended :]

Might be a good idea to add this as a comment to the installation instructions regarding what type of buttons one can or should use.

Cheers

closed 2020-02-18 by Anekd0t

laurentalacoque, 2020-02-20:
Hey @Anekd0t 
Nice you figured it out!
I'll improve the hardware button script to facilitate the discovery (one day, hopefully)