laurentalacoque/TouchSelfie-extended #91: No Handler could be found for logger auth2service

karlzre, 2020-02-26:
Hi Laurent, 

I just see that i have this issue when Photobooth try to upload a photo.
"No Handler could be found for logger auth2service"
I worked unitil monday.

Any idea ?

darrenR89, 2020-03-15:
I'm having exactly the same issue!

mattx38, 2020-04-09:
Me too !
Did someone manage fixing it ?

laurentalacoque, 2020-04-19:
Hello @karlzre 
the error "No handler could be found..." relates the the logging capability but it's only a warning and should work ok, so not the source of the problem you're experiencing.
Could you please locate your logfile (should be a file named `touchselfie.log` and paste it here or at least try to locate the source of your problem inside it?

rmlucia, 2022-05-28:
hi, 
Do you have any idea why my app is not fetching the list of photoalbums. 

I get the following error, 

"""
    Unification of email sending and album uploading using OAuth2Login
    Code for Google Photos / Picasa taken from "credentials.py" at https://github.com/wyolum/TouchSelfie
    Code for mail sending with OAuth2 taken from "apadana" at https://stackoverflow.com/a/37267330
    Assemblage and adaptation Laurent Alacoque 2o18
    
    - Update Jan 2019: Moving to photoslibrary API (Picasa API is now deprecated)
"""
import os
import base64
from apiclient import errors, discovery
from email.mime.multipart import MIMEMultipart
from email.mime.text import MIMEText
import mimetypes
from email.mime.image import MIMEImage
from email.mime.audio import MIMEAudio
from email.mime.base import MIMEBase
import webbrowser
from datetime import datetime, timedelta
from googleapiclient.discovery import build
from httplib2 import Http
from oauth2client import file, client, tools
from googleapiclient.errors import HttpError

import logging
log = logging.getLogger(__name__)

class OAuthServices:
    """Unique entry point for Google Services authentication"""
    def __init__(self, client_secret, credentials_store, username, enable_upload = True, enable_email = True, log_level = logging.WARNING):
        """Create an OAuthService provider
        
        Arguments:
            client_secret (filename) : path to an application id json file with Gmail api activated (https://console.developers.google.com)
            credentials_store (filename) : path to the credentials storage (OK to put a nonexistent file)
            username                 : gmail address of the user whom account will be used
            enable_email             : enable send_email feature
            enable_upload            : enable upload pictures feature
            log_level                : level of logging (integer, see python module logging)     
        """
        self.client_secret = client_secret
        self.credentials_store = None
        self.username = username
        self.enable_upload = enable_upload
        self.enable_email  = enable_email
        self.scopes = ""
        
        if not (self.enable_email or self.enable_upload): # if we don't want features, just return
            return 

        #build scopes
        if self.enable_upload:
            self.scopes += "https://www.googleapis.com/auth/photoslibrary.appendonly https://www.googleapis.com/auth/photoslibrary.readonly.appcreateddata "
        if self.enable_email:
            self.scopes += "https://www.googleapis.com/auth/gmail.send"
        self.scopes = self.scopes.strip()

        self.credential_store = file.Storage(credentials_store)
        
        log.setLevel(log_level)
        #mask googleapiclient info and debug messages, except in debug mode
        if (log_level == "DEBUG") or (log_level == logging.DEBUG):
            logging.getLogger("googleapiclient.discovery_cache").setLevel(logging.INFO)
            logging.getLogger("googleapiclient.discovery").setLevel(logging.INFO)
        else:
[... 308 lines cut ...]
    im.save("test_image.png")
    
    # Connecting to Google
    gservice = OAuthServices("client_id.json","storage.json",username,log_level=logging.DEBUG)

    print "\nTesting email sending..."
    print gservice.send_message(username,"oauth2 message sending works!","Here's the Message body",attachment_file="test_image.png")
    print "\nTesting album list retrieval..."
    albums = gservice.get_user_albums()
    for i, album in enumerate(albums):
        print "\t title: %s, id: %s"%(album['title'],album['id'])
        if i >= 10:
            print "skipping the remaining albums..."
            break
    print "\nTesting album creation and image upload"
    album_id = gservice.create_album(album_name="Test", add_placeholder_picture = True)
    print "New album id:",album_id
    print("Uploading to a bogus album")
    print(gservice.upload_picture("testfile.png",album_id = "BOGUS STRING" , caption="In bogus album", generate_placeholder_picture = True))
    

if __name__ == '__main__':
    test()

I do not get the list any idea or update to the api I must do?