laurentalacoque/TouchSelfie-extended #93: Pimping the interface

mattx38, 2020-04-09:
Hello,
I'm trying to "pimp" the interface to put some french touch to it ;-)

1. Can you tell me if there is a way to change the keyboard type ? I would like to use an AZERTY one instead of the QWERTY.
2. Is it possible to add a frame around the photo collage of the pictures ?

Thank you

mattx38, 2020-04-11:
I'm not sure this is the best area to make it, but I would like to suggest you some other features, to improve this PhotoBooth application.

It would be great to also have :

1. Have a preview of the image of the camera, before to launch the shots.
2. Make the screen blink betwteen shots, because it is not obvious to know when the camera takes a picture 

Otherwise, congratulations for this cool application.

mattx38, 2020-04-18:
Bonsoir Laurent, 
Pourriez-vous m'apporter des réponses ?
Je souhaiterais pouvoir terminer ce projet. Je suis de Voiron, je crois que nous ne sommes pas très loin l'un de l'autre ;-)
Merci par avance.

laurentalacoque, 2020-04-19:
Hey @mattx38 

1. To adapt the layout of the keyboard, take a look at `scripts/mykb.py`, you'll see that there already exist an `AZERTY_LAYOUT` constant that you can set as default by replacing the `layout=QWERTY_LAYOUT` on line 352.

2. A frame is already applied to collage photos, if you want to modify it, just edit the `scripts/collage_four_square.png`. It's safer to keep the same size. This image is pasted over the 2x2 collage of your photos, so remember to keep transparency wherever you want your photos to appear.

3 to 5: ... Sorry, I don't plan on implementing any new features right now, feel free to fork the project, implement them and make a pull request ;)

Finally, please avoid the nagging of reminder messages, that is just unecessary and rude.

closed 2020-04-19 by laurentalacoque

mattx38, 2020-04-19:
Ok, thank you for your answers and sorry for the inconvenience caused